OpenDevicePartnership/embedded-services #120: Create keyboard scanning and HID report generation

RobertZ2011, 2025-03-03:


jeffglaum, 2025-03-06:
@v-danielrob what's the state of this task?  Is there a (draft) PR you can link?

v-danielrob, 2025-03-06:
The PR is not ready yet.

RobertZ2011, 2025-03-27:
Use https://github.com/OpenDevicePartnership/embedded-services/blob/1646a81941200b0a6fd5c10c2ba1e42e9fc861b6/embedded-service/src/keyboard.rs#L45 as a basis for the keyboard messages. Most of that code was meant as just an example so I think what we'll need for keyboard to start with will be something like:

```
/// Keyboard event broadcast message data
#[derive(Clone)]
pub enum EventData {
    /// Actually a HID report with a standardized format
    Key([u8; 20]),
}`

/// Keyboard event message complete message
#[derive(Clone)]
pub struct Event {
    /// Source device ID
    pub from: DeviceId,
    /// Message data
    pub data: EventData,
}
```

jeffglaum, 2025-05-27:
@v-danielrob thanks for the hard work on this.  Now that you have something working, what does the timeline to get it into the ODP repo look like?  Thanks.

closed 2025-09-10 by jeffglaum