OpenDevicePartnership/embedded-services #196: Move `Controller` trait out of `embedded-usb-pd`

RobertZ2011, 2025-03-26:
Any controller trait is going to be specialized to our use cases so this trait: https://github.com/OpenDevicePartnership/embedded-usb-pd/blob/21be0b0fadecabef2ff3d9bb205591d54fb89ac8/src/asynchronous/controller.rs#L7 should be combined into this trait: https://github.com/OpenDevicePartnership/embedded-services/blob/0e41d0ba8b01c71306376564326aeb33ce6b6347/embedded-service/src/type_c/controller.rs#L168

asasine, 2025-03-27:
There are several changes across repos for this:
- [ ] Remove `PdController` trait at https://github.com/OpenDevicePartnership/embedded-usb-pd/blob/21be0b0fadecabef2ff3d9bb205591d54fb89ac8/src/asynchronous/controller.rs#L7
- [ ] Add `reset` method to https://github.com/OpenDevicePartnership/embedded-services/blob/0e41d0ba8b01c71306376564326aeb33ce6b6347/embedded-service/src/type_c/controller.rs#L168
- [ ] Implement missing `reset` method on `Tps6699x` https://github.com/OpenDevicePartnership/embedded-services/blob/0e41d0ba8b01c71306376564326aeb33ce6b6347/type-c-service/src/driver/tps6699x.rs#L35
- [ ] Remove `impl PdController for Tps6699x` in tps6699x crate https://github.com/OpenDevicePartnership/tps6699x/blob/1434684229afc9cd8c55e5d207f7e0d98f994e4d/src/asynchronous/embassy/mod.rs#L299
- [ ] Remove https://github.com/OpenDevicePartnership/tps6699x/blob/1434684229afc9cd8c55e5d207f7e0d98f994e4d/src/asynchronous/embassy/fw_update.rs#L6

asasine, 2025-03-27:
@RobertZ2011 the `embedded_usb_pd::asynchronous::controller::PdController::reset` method is only used within the `tps6699x::asynchronous::embassy::fw_update::Tps6699x::fw_update_mode_exit` method https://github.com/OpenDevicePartnership/tps6699x/blob/1434684229afc9cd8c55e5d207f7e0d98f994e4d/src/asynchronous/embassy/fw_update.rs#L60-L62

Does it need to be added to the `embedded_sevrices::type_c::controller::Controller` trait at all?

closed 2025-03-28 by jerrysxie